cmower/ros_pybullet_interface #161: Naming and message type of the topic joint_states/target

JuanDelAguila, 2022-11-21:
@joaomoura24 and I have been discussing that the topic `joint_states/target` should be renamed to `joint_states/command` to remain consistent with conventions in robot control. Additionally, the topic `joint_states/target` uses a `sensor_msgs/JointState` message type which is usually reserved for reading joint_states, not for commanding. Perhaps a more standard message type should be used for commanding joints.

cmower, 2022-11-21:
> @joaomoura24 and I have been discussing that the topic `joint_states/target` should be renamed to `joint_states/command` to remain consistent with conventions in robot control. 

There are various conventions people use, but you may be correct that "command" is more common than "target". Changing this in the interface would likely break a lot of code, and require fairly extensive fixes. I'm not convinced that this change is worthwhile. 

I suggest you consider making use of [the ros remap tag](https://wiki.ros.org/roslaunch/XML/remap) in your launch files.

> Additionally, the topic `joint_states/target` uses a `sensor_msgs/JointState` message type which is usually reserved for reading joint_states, not for commanding. Perhaps a more standard message type should be used for commanding joints. 

Again, there is several conventions people use. Potentially, we could implement a switch that the user could specify in the robot config file that tells the interface what topic types to accept. If this is a feature you need then I am very happy for you submit a PR to the interface repo in the ros_pybullet organisation.

joaomoura24, 2022-11-21:
@cmower, from my experience  I find that std_msgs/Float64MultiArray tends to be used more for commanding positions or other quantities - but we should definitely double check.
@JuanDelAguila, i think it is worth moving this issue to the official repo [ros-pybullet](https://github.com/ros-pybullet/ros_pybullet_interface).

Obviously we can map it in our own code from one type message to another and from a topic name to another.
However, i think we should consider which msg type to use according to the ones that typically ros-controllers use, to improve the ease of moving from ros-pybullet to real robot.
@cmower, if you can let us know what changes would be needed, Juan and me can try to change it in ros-pybullet.

cmower, 2022-11-21:
I think the best thing would be a flag in the config file where the user can specify the target/command message type. 

@JuanDelAguila, I will close here, please reopen this issue in the official repo and I will give you some more information on what could be implemented.

closed 2022-11-21 by cmower

JuanDelAguila, 2022-11-22:
Thank you for your comments, I will reopen this in the official repo.