cmower/ros_pybullet_interface #162: rpbi directly publishing tfs

JuanDelAguila, 2022-11-21:
The ros-pybullet interface is directly publishing tfs of the objects in the scene. @joaomoura24 and I have been discussing that it would be better to leave this to the user to implement through a `robot_state_publisher` node. It would be good at least to be able to prevent rpbi from publishing the tfs directly. 

@joaomoura24 - Please feel free to add any further comments.

cmower, 2022-11-21:
Hi @JuanDelAguila 

Thank you for opening this, you can certainly use the robot state publisher instead of the interface publishing tfs - there may be some performance benefits. To turn off broadcasting the links as tfs you can use the config parameter below 

https://github.com/cmower/ros_pybullet_interface/blob/8578b9ff798a7d4ba5766badf24769f51e2b899d/ros_pybullet_interface/src/rpbi/pybullet_robot_links.py#L26

I would like to keep this feature for the interface, and allow the user to optionally broadcast the tfs.

However, it would be great to have an example in rpbi_examples that demonstrates how to use the robot state publisher as you suggest. If you have the time please could you submit a PR to the ros_pybullet/ros_pybullet_interface repo (note, the version of the interface on my personal account won't be updated anymore, only the repositories in the ros_pybullet organisation will receive updates, so please shot any PRs and anymore issues there).

Probably, a launch file and config files that adapts one of the basic examples would be enough.

joaomoura24, 2022-11-21:
Hey @JuanDelAguila , forgot to mention that from now on, the issues should go to the [ros-pybullet](https://github.com/ros-pybullet/ros_pybullet_interface) new repo.

@cmower, I agree that it should be optional. We just didn't know how to turn it off.

JuanDelAguila, 2022-11-22:
Thank you for pointing us to that functionality to turn off broadcasting. I am a bit busy at the moment, but I will try to submit a PR with an example over the next month. In the meantime, feel free to close this.

cmower, 2022-11-22:
Thanks @JuanDelAguila

closed 2022-11-22 by cmower